RTimothyEdwards/qflow #2: Invalid verilog file detection

killruana, 2019-11-08:
qflow version 2014805d0cef7fd47b99fef2af196d1b88bffb3e (latest commit on master)

When qflow try to detect the verilog files needed for synthesizing one of my design, it incorrectly detect a "LIL" module.

This is a partial trace of the execution of the script `qflow/share/qflow/scripts/yosys.sh`

```
eval /home/killruana/opt/eda/qflow/share/qflow/bin/yosys -s k1g100.ys
grep ERROR
sed -e /\\/s#\\#/#g -e /\[/s/\[//g -e /\]/s/\]//g
set yerrcnt = `echo $yerrors | wc -c`
wc -c
echo Found cached RTLIL representation for module `$paramod/MultiplexERROR: Module `/LogicBlock' referenced in module `/LogicTile' in cell `/c_logic_block' is not part of the design.
if ( 178 > 1 ) then
set yvalid = `echo $yerrors | grep "referenced in module" | wc -c`
echo Found cached RTLIL representation for module `$paramod/MultiplexERROR: Module `/LogicBlock' referenced in module `/LogicTile' in cell `/c_logic_block' is not part of the design.
wc -c
grep referenced in module
set ymissing = `echo $yerrors | grep "is not part of" | wc -c`
echo Found cached RTLIL representation for module `$paramod/MultiplexERROR: Module `/LogicBlock' referenced in module `/LogicTile' in cell `/c_logic_block' is not part of the design.
wc -c
grep is not part of
if ( ( 178 > 1 ) && ( 178 > 1 ) ) then
set newdep = `echo $yerrors | cut -d " " -f 3 | cut -c3- | cut -d "'" -f 1`
echo Found cached RTLIL representation for module `$paramod/MultiplexERROR: Module `/LogicBlock' referenced in module `/LogicTile' in cell `/c_logic_block' is not part of the design.
cut -d   -f 3
cut -d ' -f 1
cut -c3-
set uniquedeplist =  LogicTile IOAdapter_W IOAdapter_S IOAdapter_E IOAdapter_N Multiplexer ShiftRegister LIL
else
endif
end
```

As you can see, the problem seem caused by the lack of line return character in the yosys log.

If you need to do some tests, my design is available here: https://git.slaanesh.org/killruana/kfpga-core-k1g100/src/commit/5951251936d0bf2fabee2c205cd756eb3764fc90/impl/qflow-osu035

RTimothyEdwards, 2019-11-08:
There is a quick workaround which is to create a file with the names of the files, one per line, and then put "set source_file_list=<filename>" into the project_vars.sh file.

But I will not close this issue until I have a proper fix for it.  I probably just need to redirect stdout and stderr from yosys to different places.

killruana, 2019-11-08:
Yup, I have used this solution.

RTimothyEdwards, 2019-11-08:
The real solution is that yosys.sh should use "yosys -s filename.ys > /dev/null" to look only at stderr and to remove all stdout.  I am flying today but I will apply the patch when I get home.

killruana, 2019-11-08:
Ok, thank you!

RTimothyEdwards, 2020-01-24:
Sorry, just saw this again and realized that this had not been taken care of, lost in all the busy-ness between November and the end of the year.  I just corrected the yosys.sh script to discard stdout, as discussed above, so it should no longer incorrectly flag as errors confusing things that come from the yosys normal output.

closed 2020-01-24 by RTimothyEdwards