RTimothyEdwards/qflow #4: Incorrect def output by vlog2Def when using RePlAce as the placement tool

arun13e, 2020-02-04:
I am using qflow 1.4.74 with osu035 and memory files generated by OpenRAM as hard macros, everything went smoothly with synthesis but while using RePlAce as a placement tool, RePlAce outputs `** ERROR:  Pin Instance ( dout0 ) does not exist in MACRO statement (lefPinMapStor) ` , found out that dout0 is a bus and is the output for memory macro generated by OpenRAM, vlog2def is placing buses instead of `dout0[0], dout0[1]` in the netlist part of the DEF file generated, while designs without buses seems to work fine.

RTimothyEdwards, 2020-02-04:
It does sound like vlog2def needs fixing.  Can you please make a tarball example with files and send it to me at tim@opencircuitdesign.com?

arun13e, 2020-02-04:
sure

RTimothyEdwards, 2020-02-13:
I'm going to close this issue as it has been resolved, although I'm sure there are plenty of other problems with using hard macros and/or using RePlAce, both of which have been only lightly tested in qflow.

closed 2020-02-13 by RTimothyEdwards