RTimothyEdwards/qflow #8: qflow fails with "Old format VIARULE ignored"

yurivict, 2020-03-24:
How to repeat:
1. Run ```$ placement_tool=graywolf qflow gui```
2. Choose ```Technology=osu035```
3. Choose the Verilog file ```map9v3.v```
4. Hit ```Run``` for Preparation, then for Synthesis

[image]

RTimothyEdwards, 2020-03-24:
I apologize that I am busy working on a chip tapeout and will probably not get around to looking at some of these errors in detail for a few days.

Nevertheless, pertaining to the above problem:  The "Old format VIARULE ignored" is a note, not an error (and I will put it on my to-do list that this should be printed to stdout, not stderr).  Also, I think these warnings happen earlier but are printed later because stdout and stderr are captured separately and printed separately.  The actual error that halted the flow appears to have come from graywolf ("graywolf failed with exit status 1").  If you click on the button that says "Fail", you will get a window displaying the text of the log file (which is in log/place.log), which should have more information about the error.

yurivict, 2020-03-24:
qflow looks for the graywolf command in a wrong place:
```
Qflow placement logfile created on Tue Mar 24 08:50:37 PDT 2020
Running vlog2Cel to generate input files for graywolf
vlog2Cel  -l  /usr/local/share/qflow/tech/osu035/osu035_stdcells.lef -u 100 -o /tmp/qflow/layout/map9v3.cel /tmp/qflow/synthesis/map9v3.rtlnopwr.v
No map9v3.cel1 file found for project. . . no partial blockages to apply to layout.
No map9v3.cel2 file found for project. . . continuing without pin placement hints
Running GrayWolf placement
graywolf  map9v3
/usr/local/share/qflow/bin/graywolf: Command not found.
graywolf failed with exit status 1
Premature exit.
Synthesis flow stopped on error condition.
```

I think it should look in a location where the executables are normally installed, which is ```/usr/local/bin``` in my case.

RTimothyEdwards, 2020-03-24:
No, qflow specifically keeps its own set of pointers to the applications that are known to it at the time of compiling.  So the directory /usr/local/share/qflow/bin/ contains a bunch of symbolic links to known applications.  Qflow uses this list to determine what applications it may present as available to use in the flow.

RTimothyEdwards, 2020-03-24:
Of course that raises the question of why qflow is trying to run graywolf if graywolf isn't in its list of known applications. . .

yurivict, 2020-03-24:
Without ```placement_tool=graywolf``` ```qflow``` fails: ```placement_tool: Undefined variable```

I am not sure how graywolf is supposed to be chosen in qflow.

IMO, the error messages should be more suggestive and tell the user what he is expected to do. (instead of just "Undefined variable")

yurivict, 2020-03-24:
It does create the link when graywolf is present during build.

So qflow should either just require dependencies to be present during build, or process missing dependencies during the runtime gracefully. (Personally, I would prefer the former.)

RTimothyEdwards, 2020-03-24:
Duly noted.  I only recently changed qflow to be able to use different tools and choose them from the GUI, so clearly there are a few things that need working on.  I will make sure that the error messages are clear and relevant, and see what I can do about the graceful handling during run-time as a last-ditch attempt to keep going.

76-Piyush-kudiwal, 2023-09-24:
[image]
Pls help to solve this issue.

RTimothyEdwards, 2023-09-24:
@76-Piyush-kudiwal :  As noted in my first response to the first post back in 2020, "'Old format VIARULE ignored' is a note, not an error".  Any critical error that caused the flow to fail was something else.  And, as stated in the message shown in your screenshot, the error message will be found in the `synth.log` file.

76-Piyush-kudiwal, 2023-09-24:
> @76-Piyush-kudiwal : As noted in my first response to the first post back in 2020, "'Old format VIARULE ignored' is a note, not an error". Any critical error that caused the flow to fail was [...]

@RTimothyEdwards Thanx for such quick response. But my process is stopped. Let me check once in the synth.log file.
Pls cooperate further to solve this...

76-Piyush-kudiwal, 2023-09-24:
[image]
Here, is the Snip of the .log file..

What to do now?

RTimothyEdwards, 2023-09-24:
Make sure you have installed `yosys` on your system, then rebuild/reinstall qflow so that it will have the location of yosys in its set of links.  You can also just create the link to the yosys exectuable at /usr/local/share/qflow/bin/ where all the other links are kept.

76-Piyush-kudiwal, 2023-09-24:
@RTimothyEdwards Thank You so much, It's worked !!!!!!

RTimothyEdwards, 2023-09-29:
@76-Piyush-kudiwal :  I need enough information for a fully reproducible result.

76-Piyush-kudiwal, 2023-09-29:
Okay, Pls tell what kind of info. you want. Shall I send you the pre and post sta logs ?

RTimothyEdwards, 2023-09-29:
Log files are not "enough information for a fully reproducible result".  I need all files necessary to start the STA, notably the design `.rtlnopwr.v` netlist.  I assume this is an open-source PDK like scmos or sky130?

76-Piyush-kudiwal, 2023-09-29:
Okay,I'll provide you the required files. Yes, this is an open source PDK which comes default in Qflow.

Pls, if possible can we continue our discussion on mails.
Kindly, provide me your mail ID so that I can share the files with you. 

Thank You

RTimothyEdwards, 2023-09-29:
tim AT opencircuitdesign.com (not that that's going to stop the spam bots. . .)

76-Piyush-kudiwal, 2023-09-30:
Hi @RTimothyEdwards , I've attached verilog file i.e. fa.v file ( which was simulated for Osu035 technology - dafault.)

module fa (a,b,cin,sum,co);
input a,b,cin;
output sum,co;
assign {co,sum}=a+b+cin;
endmodule

Kindly, check and update accordingly

RTimothyEdwards, 2023-10-01:
@76-Piyush-kudiwal :  Most likely this fails because of the inability of the tools to handle a 2- or 3-gate circuit.  I will try to figure out what is causing the crash, and why, though.

RTimothyEdwards, 2023-10-01:
I only had an issue with graywolf segfaulting on the Placement step.  I mitigated that by choosing an initial density of 0.1, which adds fill cells around the active logic cells.  That keeps the whole layout large enough that the tools don't get messed up by the small size.

But that doesn't sound like the error you were getting.  But any of the tools may be having issues with the small circuit size;  there may be a set of parameters where the placement succeeds but the router crashes.

76-Piyush-kudiwal, 2023-10-03:
Hi @RTimothyEdwards , I've mailed you yesterday having all the required files attached as you mentioned. Kindly, check. 

Thank You

76-Piyush-kudiwal, 2023-12-22:
Hi @RTimothyEdwards , I wanted to ask that I'm trying picorv32.v and map9v3.v  source files at  45nm. But, facing error in STA and  DRC,  moreover I've also tried other designs at same node. So, Is there any solution for this to run at 45nm?. Have you tried any design at  45nm.
Kindly reply.

Thank You

RTimothyEdwards, 2023-12-22:
No, I only use PDKs that correspond to real foundry processes, and there is no such PDK for 45nm.  If you're using the gscl45nm tech files in qflow, though, I don't recall any particular issues with them.

76-Piyush-kudiwal, 2023-12-22:
Okay, Thank You @RTimothyEdwards  for the quick reply.

76-Piyush-kudiwal, 2023-12-22:
Hi @RTimothyEdwards,  If I want to use PDK  for lower technology node other than 45nm, then is it possible to add it in Qflow and pls tell how can I add that PDK in the Qflow.

Kindly reply.

Thank You

RTimothyEdwards, 2023-12-22:
Honestly, if you want to do anything smaller than 90nm feature size, you should abandon qflow completely and use OpenROAD instead.  Qflow was never designed for anything large-scale or small feature size (although at one point I started integrating OpenROAD tools, but OpenROAD development outstripped my ability to keep up with it and I haven't gotten back to the task).  OpenROAD was designed specifically for nodes like 12, 14, 16nm;  and although it is being used for (and works quite well with) nodes like 130nm and 180nm, that was not its main development focus.

76-Piyush-kudiwal, 2023-12-23:
Okay, Okay. Now, I get it. 
Thank You for this detailed reply.