RTimothyEdwards/qflow #10: Error using with RePlAce. `ntuplace3: too many levels of symbolic links`

psgff, 2020-04-29:
Hello! I'm trying to use qFlow with RePlAce but receiving error. Could you please help? I've taken RePlAce from their releases, because didn't manage to build it on my machine. I'm using Ubuntu 18.04.
```
INFO:  SUCCESSFULLY WRITE DEF FILE INTO outputs/ispd/fle_preplace/experiment000/fle_preplace_gp.def
ln -s /usr/local/share/qflow/bin/ntuplace3 outputs/ispd/fle_preplace/experiment000/tiers/0/
[INFO] Call NtuPlacer3
cd outputs/ispd/fle_preplace/experiment000/tiers/0 && ./ntuplace3 -aux fle_preplace.aux -loadpl 

fle_preplace.pl -util 1.00 -noglobal 

sh: 1: ./ntuplace3: Too many levels of symbolic links
INFO:  READ AGAIN: outputs/ispd/fle_preplace/experiment000/tiers/0/fle_preplace.ntup.pl
ERROR: outputs/ispd/fle_preplace/experiment000/tiers/0/fle_preplace.ntup.pl is not Exist! 
Aborting !! 
RePlAce failed with exit status 1
Premature exit.
Synthesis flow stopped on error condition.
```

RTimothyEdwards, 2020-04-29:
The symbolic link levels error comes from a shell script and suggests that there is a symbolic link pointing to itself which is causing an infinite recursion.  The origin of the loop should be at (depending on where you installed qflow) /usr/local/share/qflow/bin/ntuplace3.  On my system, this is a link to /usr/local/bin/ntuplace3 (which comes from compiling and installing RePlAce from source, I think).

Ultimately, what I need to do is to update qflow with the newer incarnation of OpenROAD as a single executable driven by Tcl commands.  The current implementation in qflow is from a much earlier version of OpenROAD, when all the tools were separate.

psgff, 2020-04-29:
Thank you for a quick reply! Yes, definitely it's the problem. Pointing to itself. May be meanwhile you could point me the version of RePlAce, that you've used? I tried standalone branch in their repo, but it fails and I'm completely stuck.

RTimothyEdwards, 2020-04-29:
What I have, I got by doing `git clone --recursive https://github.com/abk-openroad/RePlAce.git`, and (maybe?) following the instructions in README.md.  There's a CentOS install script (which I probably used for my Fedora system) and an Ubuntu install script.  If it's any additional help, the version I compiled was commit ffee1809325d780509447c196151e7f7df647107 from about a year ago.

psgff, 2020-04-29:
Unfortunately this repo is dead. I'll try to investigate the new one... (https://github.com/The-OpenROAD-Project/RePlAce)

psgff, 2020-04-30:
I've managed to make ntuplace3 work. sort of... Checked out the main repo as `ffee1809325d780509447c196151e7f7df647107`, make failed but at least ntuplace3 was there, so I've linked it to the qflow bin dir and the error is gone. However there is another error now, going to find out what does it mean.
```
[STAGE 2]: Legalization (circuit = fle_preplace.aux) MEM= 35 MB

------------------------------------------------------------------------------------*******************legalization dis_value 26432.36

# Average displacement per cell:         33.50% BBox
# Max displacement of the cell:          82.22% BBox
# Cell number with > 1.5% BBox movement: 77.54%
# HPWL change:                           24.21%
# Average net length change per net:     6.07% BBox
# Max net length change of the net:      79.05% BBox

Legalization failed
Unlegal count: 286
INFO:  READ AGAIN: outputs/ispd/fle_preplace/experiment000/tiers/0/fle_preplace.ntup.pl
ERROR: outputs/ispd/fle_preplace/experiment000/tiers/0/fle_preplace.ntup.pl is not Exist! 
Aborting !! 
RePlAce failed with exit status 1
Premature exit.
Synthesis flow stopped on error condition.
```

praharshapm, 2020-08-06:
@psgff did you manage to install the standalone repo of RePlAce?