RTimothyEdwards/qflow #13: Include Spice Models

belohoub, 2020-06-29:
First, thank you for your great work on qflow, Magic, ... !

I resurrected the SPICE model collection originated in ancient MOSIS MPW runs. Some of the models are related to qflow/OSU technology used by qflow. These models are nice-to-have for MC simulations and experiments with real technology node. Up today it's probably the most complete online collection of these models - the original MOSIS webpage was shut-down and people are normally using one or two model files (as examples downloaded for university courses some time ago).

If you know about other collections, please, let me know, I'm currently missing about 2/3 of original MOSIS models.

My collection resides here: https://github.com/DDD-FIT-CTU/CMOS-SPICE-Model-Collections

If you find it useful, please, consider including it to qflow.

RTimothyEdwards, 2020-06-29:
The lack of SPICE models is a known issue, and I was not aware that MOSIS posted any device models, or that any of them were considered "open".  I'll take what I can from your data dump.  I also have a dump from the now-extinct MOSIS ftp site that I can look through to see if I have any additional files.  I don't think so, though.

I'm kind of busy at the moment (with this:  https://fossi-foundation.org/2020/06/17/fossi-dial-up) but if see no action on this after a week, please remind me to take care of it.

belohoub, 2020-06-30:
Thank you! And thank you for pointing me to that event - I'll try to join you online!

Right, this is still a serious issue. The repository currently provides "some idea" about real parameter variations in normal production (years ago), however, exact corner models are still missing. I think, that it remains a serious issue. In the "open world", I saw some corner models derived for OSU 45nm from [PTM](http://ptm.asu.edu/), but nothing more.

Years ago, MOSIS provided test reports for many MPW runs (maybe to all, I don't know) including characterized SPICE models. I found no license information, so I suggest it is just a "report provided as-is". At least for non-commercial use, it should be ok. I guess, that my license note in the repository should be ok.

Some of these model files circulate on the web and these are used as "spice model examples" (materials for students, etc.; one of them is also included in the OSU lib!). These are normally (de-facto) considered as nominal models ... but this is probably a mistake ...

If you would find something more, It would be great, I'm curious! :-)

The models resided originally in the "/Technical/Testdata/" path of the mosis webpage - the FTP could have a similar structure; some notest to the particular collections are in my repository, here e.g. for [TSMC180nm](https://github.com/DDD-FIT-CTU/CMOS-SPICE-Model-Collections/tree/github/MOSIS_waferTestData_TSMC_180nm)

belohoub, 2020-07-20:
I'm back to remind me :-) Heve you tried to find something?

belohoub, 2020-08-09:
@RTimothyEdwards  Have you tried to find somethink?

And just for curiosity: will be SPICE corner models available for SkyWater sky130A?

RTimothyEdwards, 2020-08-09:
Yes, there will be a full set of SPICE corner models for sky130A.

belohoub, 2020-08-09:
Thank you! ... that's ... terrific! :-)