RTimothyEdwards/qflow #16: running through map9v3 tutorial after installing qflow and all the dependencies.  getting error "migrate_tool undefined variable"

donjonson, 2020-12-15:
I did not yet have migic installed when I installed qflow but I do now.  however every time I run qflow it generates the project_vars.sh with many variables left empty. including migrate_tool, lvs_tool, drc_tool, gds_tool, display_tool.

I tried filling it out manually with the defaults indicated on the webesite magic_db etc.  that allowed me to succed on the intiliazation step however with the synthesis step it apparently regenerates the project_vars.sh again without those variables defined. 

how do I configure qflow to define thes? or at least not overwrite them when I manually define them?

donjonson, 2020-12-15:
I see the problem:  in the qflow.sh script there are if(0) for this:  I will put if(1) and see what happens

#---------------------------------------------------------
# Make sure that the flow tools are defined with defaults
#---------------------------------------------------------

if ( ! (${?synthesis_tool}) ) then
   if ( 1 ) then
      set synthesis_tool=yosys
   endif
endif

if ( ! (${?placement_tool}) ) then
   if ( 1 ) then
      set placement_tool=graywolf
   else if ( 0 ) then
      set placement_tool=replace
   endif
endif

if ( ! (${?sta_tool}) ) then
   set sta_tool=vesta
endif

if ( ! (${?router_tool}) ) then
   if ( 1 ) then
      set router_tool=qrouter
   endif
endif

if ( ! (${?migrate_tool}) ) then
   if ( 0 ) then
      set migrate_tool=magic_db
   endif
endif

if ( ! (${?lvs_tool}) ) then
   if ( 0 ) then
      set lvs_tool=netgen_lvs
   endif
endif

if ( ! (${?drc_tool}) ) then
   if ( 0 ) then
      set drc_tool=magic_drc
   endif
endif

if ( ! (${?gds_tool}) ) then
   if ( 0 ) then
      set gds_tool=magic_gds
   endif
endif

if ( ! (${?display_tool}) ) then
   if ( 0 ) then
      set display_tool=magic_view
   endif
endif

RTimothyEdwards, 2020-12-15:
@donjonson :  These "1" and "0" values are generated from the results of running "configure", so the apparent behavior is that on doing "configure", the script did not find the location of most of the tools (qrouter was found, but magic, netgen, and graywolf were not).  Since the script did not find them automatically, then the path to the tools would need to be passed to the "configure" script.

RTimothyEdwards, 2020-12-15:
Correction:  graywolf was found (also yosys was found).  Only magic and netgen are not found.

donjonson, 2020-12-19:
Thank you for the help I was able to get it working!  well at least that much.

closed 2020-12-21 by donjonson