RTimothyEdwards/qflow #17: running QFLO with new technology. M1 routes are shorting to standard cell M1 shapes

donjonson, 2020-12-19:
Hello Thanks for this open source software.  this is probably more of a configuration question than a bug report so I am hoping this is the right place.

with exception of the magic tech file I have made a tech directory for the qflow digital flow.  it seems to be working up to succesful rout and it dies because it does not create an rc file.  but that is not the reason for my question here.  I am asking what file I need to modify so tha tthe routes do not short to the standard cells.  is it a placement "*.par"  problem? or is it some configuration I need to setup for the router?

please see the screenshot below with blue routes generated by qflow shorting to the blue M1 metal within the standard ce..
 
[image]

RTimothyEdwards, 2020-12-20:
This looks like an offset error.  Note that in your screenshot, the routes are all on the grid, but the pins are all off-grid;  looks like exactly 1/2 grid off.  If you pulled all of the standard cells down and to the right by 1/2 grid pitch, it looks like everything is routed perfectly.  That is, qrouter is seeing the correct layout and producing a corresponding routed result, but the import of the routed DEF file has shifted the cell positions.  Probably this has to do with the origin in the LEF view vs. the origin in the GDS view, but I'm not sure.  Is the above view from reading GDS and displaying only the metal layers, or is it from reading DEF?

donjonson, 2020-12-20:
Thanks for your response.  This is a result of reading in the def file into my layout editing cad tool.  I tried shifting it and it is still not quite right.  it does seem like qrouter is seeing the standard cells because I can make sense of what it is trying to do but it seems there is a scale and or grid off.  

 One thing I can tell you is that it is not clear to me how to communicate to qrouter what the parameters are for routing.  is it in the .par file that graywolf reads?  If so I have been messing with that trying to affect the routing and it has but I am not sure what values to put in there to get it to route correctly.   I think the foundry provided all the config files qrouter wanted except for the techname.sh that .par file.  So I took an educated guess on what goes in there.  

here is what I put.  (I am sure it is wrong but not sure how it affects the routing)

I basically have two routing metals. M1 and MT.  m1 has a min width of 0.16 um with a min space of 0.16 um and MT is 0.2um width with 0.2um space.  I am not sure what the routing grid should be. 

# NOTE:  all distance units are in centimicrons unless otherwise stated

`RULES
    # values are resistance in ohms/sq and capacitance in fF/um^2
    layer M1 0.07 0.030 horizontal
    layer MT 0.07 0.006 vertical

    via V1 M1 MT

    width M1 16 
    width MT 20
    width V1 20

    # Set spacing = track pitch - width, so that GrayWolf places pins
    # on the right pitch.
    # Pitches are (in um):

    spacing M1 M1 160
    spacing MT MT 200

    # Stacked vias allowed

    overhang V1 M1 4
    overhang V1 MT 0

ENDRULES

*vertical_wire_weight : 1.0
*vertical_path_weight : 1.0
*padspacing           : variable
*rowSep		      : 0.0   0
*track.pitch	      : 1
*minimum_pad_space    : 1
*gridX		      : 1
*gridY		      : 1
*gridOffsetX	      : 0
*gridOffsetY	      : 0
*graphics.wait        : off
*last_chance.wait     : off
*random.seed	      : 12345

TWMC*chip.aspect.ratio : 1.0

TWSC*feedThruWidth    : 80 layer 1
TWSC*do.global.route  : on
TWSC*ignore_feeds     : true
TWSC*call_row_evener	: true
TWSC*even_rows_maximally : true
# TWSC*no.graphics    : on

GENR*row_to_tile_spacing: 1
# GENR*numrows		: 6
GENR*flip_alternate_rows : 1`

RTimothyEdwards, 2020-12-20:
Most of the values in the .par file are ignored, because the values are taken from the technology LEF file and other sources.  If you have a technology LEF file, that's the best thing to use.  The .par file is only meaningful to the placement tool, anyway.  The fact that it looks like qrouter routed something suggests that you have, and are using, a valid technology LEF file.

donjonson, 2020-12-20:
I am sorry to keep bothering you.  I know you have provided a free tool and now you are providing free support too!  when I have annoyed you enough just tell me to take a hike :).  yes the tech.lef does have all of the correct information in there. But something seems to be off in the way it is routing.  I am not sure if it is a scaling problem or an offset.  but the routing grid does not seem to match the pins in the standard cells.  and also the M1 routes seem to avoid some of the standard cell metal but short through some of it too.  also it seems to ignore minimum spacing rules.   it almost seems like it expects a larger gap between standard cells. or something like that. if I shift the cells over to make the pins on routing grid it only fixes some of them.  And see how the selected route seems to ignore and route through the M1 in the center cell? I am sure this is simply a configuration problem. scaling, spacing etc... but I am just not smart enough to figure it out.  it seems that perhaps the routing grid may supposed to be differente between x and y.  is that possible?

[image]

this experiment is a very simple inverter and nand gate with a buffer just so I can make sense of it.

donjonson, 2020-12-20:
sorry last comment.  Here is another example where it is shorting power and ground together.  
also if you look at the routes it seems like if you shrank the standard cells the route may actually work out. 

and for some reason one of the pins is being placed almost a milimeter off to the right. 

[image]

1347806, 2020-12-28:
I'm very curious which fab lib file you use? lib with qlow package or lib from other FAB vendor?

donjonson, 2020-12-28:
Hi I am using a current release of a major US fab. I am a little reluctant
to give specifics due to nda's but it is a well established USfab with a
long history in the industry.  I believe the kit is geared towards the
cadence tool flow.  I am certain the pdk files are as rock solid as any
pdk.  A likely problem here is that I have next to zero experience with
digital place and route and dont fully understand how to properly configure
.  It looks to me that the router is only seeing pin shapes in the standard
cells and not the metal one shapes without pins.  but that is just how it
appears to me and I may certainly be wrong.

I forked the qflow project so I could try to dig into the code to see what
I am doing wrong but I havn't had a chance to do that yet.  honestly the
placement that graywolf looks good and I was going to just try and route it
by hand.  it is not that big of a circuit.