RTimothyEdwards/qflow #19: qflow aborts on error in magic script during migration step

svenn71, 2021-02-01:
This happens both with qflow gui and command line:
```
Qflow migration logfile created on Sun 31 Jan 2021 10:04:43 PM CET
magic -dnull -noconsole  migrate_map9v3.tcl
/usr/local/share/qflow/bin/magic: line 18: cd: /usr/local/share/qflow/bin/../lib/magic/tcl: No such file or directory
/usr/local/share/qflow/bin/magic: line 66: /home/svennb/work/eda/qflow/map9v3/layout/magicdnull: No such file or directory
magic failed with exit status 127
Premature exit.
Synthesis flow stopped on error condition.
```

svenn71, 2021-02-02:
This is a result of a symlink dereferencing from `/usr/local/share/qflow/bin/magic` to `/usr/local/bin/magic` the `else`-clause in `magic.sh` not quite doing its job. In `magic.sh`:
```
TCL_REL_OR_ABS=../lib/magic/tcl
if [ "${TCL_REL_OR_ABS:0:1}" = "/" ]; then
   TCL_DIR=$TCL_REL_OR_ABS
else
   TCL_DIR=$(cd $(dirname ${BASH_SOURCE[0]})/$TCL_REL_OR_ABS; pwd -P)
fi
```
Since I have compiled all qflow tools from git and installed into `/usr/local`, a quick-fix is to set
```
TCL_REL_OR_ABS=/usr/local/lib/magic/tcl
```
in `/usr/local/bin/magic`

I am not sure if this is a `magic` problem or a `qflow` problem.

RTimothyEdwards, 2021-02-02:
Not sure either.  The code that seems to be the problem was added from a pull request and was supposed to help make magic more relocatable.  But it seems to break qflow.  It assumes that the launch script "magic" is called from its install location and cannot tolerate it being called from a symbolic link.  Presumably the fix is to first dereference the symbolic link.  So I would call that an error in magic.

RTimothyEdwards, 2021-02-02:
Thank you for pointing this out.  Pull request #41 in the magic repo was responsible for breaking the ability to call magic from a symbolic link as is done in qflow.  I have reimplemented the code in the pull request so as to preserve the intent but fix the usage in qflow.  If you update to magic `commit c0867ef71e46274b397b253bbcb0d9631d5e2d15` (master branch, opencircuitdesign.com) it should correct the error in qflow.