RTimothyEdwards/qflow #20: vlog2Cel being invoked with incorrect file input

kinjav03, 2021-02-02:
I'm doing a run-through of qflow 1.4 with the `map9v3.v` file used as an example, and the flow stops at the placement step since vlog2Cel is being pointed at `.rtlnopwr.v`, rather than `map9v3.rtlnopwr.v` which was appropriately generated. Here's the output of `qflow place map9v3`:

```--------------------------------
Qflow project setup
--------------------------------

Technology set to osu035 from existing qflow_vars.sh file
Regenerating files for existing project map9v3
Qrouter detail maze router version 1.4.83.T
Reading LEF data from file /usr/local/share/qflow/tech/osu035/osu035_stdcells.lef.
LEF Read, Line 135: NOTE:  Old format VIARULE ignored.
LEF Read, Line 136: NOTE:  Old format VIARULE ignored.
LEF Read, Line 140: NOTE:  Old format VIARULE ignored.
LEF Read, Line 141: NOTE:  Old format VIARULE ignored.
LEF Read, Line 151: NOTE:  Old format VIARULE ignored.
LEF Read, Line 152: NOTE:  Old format VIARULE ignored.
LEF Read, Line 156: NOTE:  Old format VIARULE ignored.
LEF Read, Line 157: NOTE:  Old format VIARULE ignored.
LEF Read, Line 167: NOTE:  Old format VIARULE ignored.
LEF Read, Line 168: NOTE:  Old format VIARULE ignored.
LEF Read, Line 172: NOTE:  Old format VIARULE ignored.
LEF Read, Line 173: NOTE:  Old format VIARULE ignored.
LEF file:  Defines site corner (ignored)
LEF file:  Defines site IO (ignored)
LEF file:  Defines site core (ignored)
LEF read: Processed 3179 lines.
LEF Read: encountered 0 errors and 12 warnings total.
Vertical route layer at non-minimum pitch 3.2.  Using smaller pitch 1.6, will route on 1-of-2 tracks for layer metal4.
Running vlog2Cel to generate input files for graywolf
vlog2Cel  -l  /usr/local/share/qflow/tech/osu035/osu035_stdcells.lef -u 100 -o /home/linda/projects/hardware/map9v3/layout/.cel /home/linda/projects/hardware/map9v3/synthesis/.rtlnopwr.v
Segmentation fault
vlog2Cel failed with exit status 139
Premature exit.
```

RTimothyEdwards, 2021-02-02:
How did you invoke qflow?  It appears that the script `graywolf.sh` got an empty string for the 2nd argument, and so set `$rootname` to an empty string.  Apparently this happened through several steps, because the script did not fail on checking the existence of file `${rootname}.info`, so apparently an earlier script had an empty string argument and left a file `.info` in the layout directory.

kinjav03, 2021-02-02:
The same problem occurs when invoking qflow from the command line as described by the tutorial (`qflow synthesize place route map9v3`,) as well as running the `place` step by itself or running through the flow via the GUI. It also seems to persist even after working from a clean project directory.

kinjav03, 2021-02-06:
I came back to this problem today, removing the `:h` in `set rootname=${sourcename:h}` where it's used fixes the problem (I'm using zsh to invoke these scripts, if that has anything to do with it.) I'd like to find a less hacky solution, though.

RTimothyEdwards, 2021-02-06:
Odd. . . the 1st line invokes tcsh;  the scripts are all in tcsh syntax.  And the tcsh man page clearly states that history substitutions are allowed in variable substitutions, so `${sourcename:h}` should be the head of path `${sourcename}`.  Removing the `:h` works as long as qflow is called with the root name of a project;  perhaps that is the only way it's ever called?

ntdesmond, 2021-08-09:
I've run into the same problem on qflow 1.3 in [placement.sh](https://github.com/RTimothyEdwards/qflow/blob/qflow-1.3/scripts/placement.sh). Strangely enough, it worked fine before, but now does not. I believe I didn't do any changes between runs.
However, I don't find a reason it can/could work successfully since what `${sourcename:h}` returns is actually a directory of a path, however, according to default `qflow_exec.sh` the `sourcename` is set to `map9v3` from the second argument (checked both exec scripts for the run that succeeded long ago and the one that does not succeed now). Thus, `${sourcename:h}` returns an empty string every time. Again, I don't have an idea why it worked successfully before.