RTimothyEdwards/qflow #21: "set: No match." error in yosys.sh

xphoniex, 2021-03-01:
I'm trying to synthesize this [processor](https://github.com/bluespec/Piccolo), specifically this [build](https://github.com/bluespec/Piccolo/tree/master/builds/RV64ACDFIMSU_Piccolo_iverilog).

My flow is failing right at **Synthesis** with message:
```
$ ./qflow_exec.sh
set: No match.
```

After diving into `yosys.sh`, the culprit seems to be [yosys.sh#L379](https://github.com/RTimothyEdwards/qflow/blob/master/scripts/yosys.sh#L379). If I add this line before that:
```
   ${bindir}/yosys -s ${modulename}.ys |& sed -e "/\\/s#\\#/#g" -e "/\[/s/\[//g" -e "/\]/s/\]//g" | grep ERROR
   # Note:  Remove backslashes and brackets to avoid problems with tcsh
   set yerrors = `eval ${bindir}/yosys -s ${modulename}.ys |& sed -e "/\\/s#\\#/#g" \
                -e "/\[/s/\[//g" -e "/\]/s/\]//g" | grep ERROR`

...
```
and run my script again, my output becomes:

```
$ ./qflow_exec.sh
ERROR: Module `/FIFO20' referenced in module `/mkCPU' in cell `/stage3_f_reset_rsps' is not part of the design.
ERROR: Module `/mkRISCV_MBox' referenced in module `/mkCPU' in cell `/stage2_mbox' is not part of the design.
ERROR: Module `/mkFBox_Top' referenced in module `/mkCPU' in cell `/stage2_fbox' is not part of the design.
ERROR: Module `/mkSoC_Map' referenced in module `/mkCPU' in cell `/soc_map' is not part of the design.
ERROR: Module `/mkNear_Mem' referenced in module `/mkCPU' in cell `/near_mem' is not part of the design.
ERROR: Module `/mkGPR_RegFile' referenced in module `/mkCPU' in cell `/gpr_regfile' is not part of the design.
ERROR: Module `/mkFPR_RegFile' referenced in module `/mkCPU' in cell `/fpr_regfile' is not part of the design.
ERROR: Module `/FIFO2' referenced in module `/mkCPU' in cell `/f_reset_rsps' is not part of the design.
ERROR: Module `/mkCSR_RegFile' referenced in module `/mkCPU' in cell `/csr_regfile' is not part of the design.
ERROR: Module `/mkCSR_MIP' referenced in module `/mkCSR_RegFile' in cell `/csr_mip' is not part of the design.
ERROR: Module `/mkCSR_MIE' referenced in module `/mkCSR_RegFile' in cell `/csr_mie' is not part of the design.
ERROR: Module `/RegFile' referenced in module `/mkFPR_RegFile' in cell `/regfile' is not part of the design.
ERROR: Module `/mkMMU_Cache' referenced in module `/mkNear_Mem' in cell `/icache' is not part of the design.
use of @* instead of @(...) for better match of ERROR: Module `/mkFBox_Core' referenced in module `/mkFBox_Top' in cell `/fbox_core' is not part of the design.
set: No match.
```

This seems a problem on part of `tcsh` and can(?) be solved if we stop piping stderr so `|&` becomes `|`, yet I'm still not able to synthesize because `_mapped.v` file does not exist. Full output after change:

```
$ ./qflow_exec.sh
ERROR: Module `\FIFO20' referenced in module `\mkCPU' in cell `\stage3_f_reset_rsps' is not part of the design.
ERROR: Module `\FIFO20' referenced in module `\mkCPU' in cell `\stage3_f_reset_rsps' is not part of the design.
Running yosys for verilog parsing and synthesis
yosys  -s mkCPU.ys

 /----------------------------------------------------------------------------\
 |                                                                            |
 |  yosys -- Yosys Open SYnthesis Suite                                       |
 |                                                                            |
 |  Copyright (C) 2012 - 2016  Clifford Wolf <clifford@clifford.at>           |
 |                                                                            |
 |  Permission to use, copy, modify, and/or distribute this software for any  |
 |  purpose with or without fee is hereby granted, provided that the above    |
 |  copyright notice and this permission notice appear in all copies.         |
 |                                                                            |
 |  THE SOFTWARE IS PROVIDED "AS IS" AND THE AUTHOR DISCLAIMS ALL WARRANTIES  |
 |  WITH REGARD TO THIS SOFTWARE INCLUDING ALL IMPLIED WARRANTIES OF          |
 |  MERCHANTABILITY AND FITNESS. IN NO EVENT SHALL THE AUTHOR BE LIABLE FOR   |
 |  ANY SPECIAL, DIRECT, INDIRECT, OR CONSEQUENTIAL DAMAGES OR ANY DAMAGES    |
 |  WHATSOEVER RESULTING FROM LOSS OF USE, DATA OR PROFITS, WHETHER IN AN     |
 |  ACTION OF CONTRACT, NEGLIGENCE OR OTHER TORTIOUS ACTION, ARISING OUT OF   |
 |  OR IN CONNECTION WITH THE USE OR PERFORMANCE OF THIS SOFTWARE.            |
 |                                                                            |
 \----------------------------------------------------------------------------/

 Yosys 0.7 (git sha1 61f6811, gcc 6.2.0-11ubuntu1 -O2 -fdebug-prefix-map=/build/yosys-OIL3SR/yosys-0.7=. -fstack-protector-strong -fPIC -Os)

-- Executing script file `mkCPU.ys' --
[... 185 lines cut ...]
Note: Assuming pure combinatorial block at /.../qflow-piccolo/source/mkCPU.v:5601 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Note: Assuming pure combinatorial block at /.../qflow-piccolo/source/mkCPU.v:5616 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Note: Assuming pure combinatorial block at /.../qflow-piccolo/source/mkCPU.v:5651 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Note: Assuming pure combinatorial block at /.../qflow-piccolo/source/mkCPU.v:5663 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
Successfully finished Verilog frontend.

3. Executing SYNTH pass.

3.1. Executing HIERARCHY pass (managing design hierarchy).

3.1.1. Analyzing design hierarchy..
ERROR: Module `\FIFO20' referenced in module `\mkCPU' in cell `\stage3_f_reset_rsps' is not part of the design.
outputprep failure:  No file mkCPU_mapped.v.
Premature exit.
```

**Any help would be appreciated.**

xphoniex, 2021-03-03:
So inside `tcsh` running:

```
> /usr/local/share/qflow/bin/yosys -s mkCPU.ys 

.
.
.
.
.

Note: Assuming pure combinatorial block at mkMMU_Cache.v:4692 in
compliance with IEC 62142(E):2005 / IEEE Std. 1364.1(E):2002. Recommending
use of @* instead of @(...) for better match of synthesis and simulation.
ERROR: Module `\mkFBox_Core' referenced in module `\mkFBox_Top' in cell `\fbox_core' is not part of the design.
```

but somehow `grep` modifies it a bit:

```
> /usr/local/share/qflow/bin/yosys -s mkCPU.ys | & grep ERROR
use of @* instead of @(...) for better match of ERROR: Module `\mkFBox_Core' referenced in module `\mkFBox_Top' in cell `\fbox_core' is not part of the design.
```

_(where did the `synthesis and simulation.` part go?)_

Anyway I kinda cheated by removing anything extra before 'ERROR' and it works now:

```
set yerrors = `eval ${bindir}/yosys -s ${modulename}.ys |& sed -e "/\\/s#\\#/#g" \
                -e "/\[/s/\[//g" -e "/\]/s/\]//g" | grep ERROR | sed 's/^.*\(ERROR.*\).*$/\1/'`
```

Can you please take a look @RTimothyEdwards to make sure this doesn't break anything else? I can make a patch.

RTimothyEdwards, 2021-03-05:
Honestly, I think the answer to annoying issues with tcsh and other annoying issue with bash is to stop using shell scripts altogether and rewrite everything in python.

The problem here is that "|&" does not, as I would like, direct only stderr to the pipe;  you can only either redirect stdout or both stdout and stderr together.  Because they are output together but buffered separately, there is no guarantee that stderr and stdout won't be interleaved.  Really, the best solution is to do it in python, where one can call subprocess() and then handle the stdout and stderr streams separately.

Your solution seems reasonable, so I'll adopt it for now.