RTimothyEdwards/qflow #22: Issue in some scripts with usage of "set" command

jscatena88, 2021-04-01:
I was receiving an error message in the GUI trying to run the "Migration" step, "set: Variable name must begin with a letter". Poking round it seems that line 169 in magic_db.sh has the issue. Specifically changing 
`set techfile ${techdir}/${techfile}` 
to  
`set techfile=${techdir}/${techfile}`
fixed the issue. I believe there is a similar issue in magic_drc.sh line 150

jscatena88, 2021-04-01:
also in magic_gds.sh 

seems to stem from the 2/22 commit "Modified the way that the magic scripts and the manager GUI handle "

K0rrid0r, 2021-04-03:
[image]

I'm having the same problem.

I changed:
magic_db.sh
magic_gds.sh
magic_drc.sh

to:
set techfile=${techdir}/${techfile}

but it didn't move past Migration
Any other suggestion?

ju-sh, 2021-04-11:
@jscatena88 Thanks for telling us about this modification. Was able to get past migration with those changes.

Now stuck at LVS just like @K0rrid0r mentions at #23 .

svenn71, 2021-08-12:
diff --git a/scripts/magic_db.sh b/scripts/magic_db.sh
index 7edda7e..5f6624c 100755
--- a/scripts/magic_db.sh
+++ b/scripts/magic_db.sh
@@ -166,7 +166,7 @@ set subv=`echo $version | cut -d. -f3`
 
 if ( !(-r $techfile)) then
    if (`echo $techfile | cut -c1` != "/") then
-      set techfile ${techdir}/${techfile}
+      set techfile=${techdir}/${techfile}
    endif
 endif
 if (-r $techfile) then
diff --git a/scripts/magic_drc.sh b/scripts/magic_drc.sh
index 8f9b73e..0f98b89 100755
--- a/scripts/magic_drc.sh
+++ b/scripts/magic_drc.sh
@@ -147,7 +147,7 @@ endif
 
 if ( !(-r $techfile)) then
    if (`echo $techfile | cut -c1` != "/") then
-      set techfile ${techdir}/${techfile}
+      set techfile=${techdir}/${techfile}
    endif
 endif
 if (-r $techfile) then
diff --git a/scripts/magic_gds.sh b/scripts/magic_gds.sh
index 2bf7abe..6b067f2 100755
--- a/scripts/magic_gds.sh
+++ b/scripts/magic_gds.sh
@@ -119,7 +119,7 @@ set subv=`echo $version | cut -d. -f3`
 
 if ( !(-r $techfile)) then
    if (`echo $techfile | cut -c1` != "/") then
-      set techfile ${techdir}/${techfile}
+      set techfile=${techdir}/${techfile}
    endif
 endif
 if (-r $techfile) then
diff --git a/scripts/magic_view.sh b/scripts/magic_view.sh
index 8671f4b..3b8868b 100755
--- a/scripts/magic_view.sh
+++ b/scripts/magic_view.sh
@@ -199,7 +199,7 @@ set dispfile="${layoutdir}/load_${rootname}.tcl"
 # absolute path.
 if ( !(-r $techfile)) then
    if (`echo $techfile | cut -c1` != "/") then
-      set techfile ${techdir}/${techfile}
+      set techfile=${techdir}/${techfile}
    endif
 endif
 
﻿

jscatena88, 2023-02-23:
This issue appears to be fixed by this commit: https://github.com/RTimothyEdwards/qflow/commit/a2d468e5ef3c748fb2df2c6baa6287ad0fdcfcac

Closing the issue