RTimothyEdwards/qflow #23: LVS fail - Errors encountered in synthesis flow

K0rrid0r, 2021-04-06:
[image]

Just trying to run the test seems impossible at this point, tried several different commits all fail at some point.

This is commit f3c9713f527ab068caaeaf269f387e156e9d73c1 and its the closest ive come to the end step Cleanup

Any suggestions how to fix this?, I just want to run the test successfully :)

ju-sh, 2021-04-11:
I too am having this problem. Hopefully a solution can be found.

ju-sh, 2021-04-11:
But my error is slightly different from that of @K0rrid0r.

For me it says:

    Circuit 1 contains 206 devices,  Circuit 2 contains 206 devices
    Circuit 1 contains 228 nets,     Circuit 2 contains 220 nets

svenn71, 2021-07-12:
On one machine where I can complete qflow gui with map9v3 without errors, sha 08ffd272bcc07c63a831477c589f9b568b9f2ea5 of qflow is installed. A git diff origin/master show loads of changes, among them, the failing set commands mentioned.

S-PADMANABAN-2001, 2022-11-14:
Try to make the codes given inside always@ static