RTimothyEdwards/qflow #25: "migration" step fail

PiyushSaini97, 2021-06-17:
Hi-

I have been trying to synthesize a basic full adder using qflow. However, when I run the software, the process encounters an error at "migration" step. You can see in the attached image
[image]

I don't understand what the error is about. None of the variables I used contain a number or special character.
I'm also attaching the verilog code I'm trying to synthesize.
[image]

Please help.
Thank you.

Regards,
Piyush

ThimayaPR, 2021-07-02:
Hello, so I had the exact same error and tried some debugging through the past few days but to no avail. 

It seems like a tcsh syntax error but i couldn't track it down, the changes i  made to both the _qflow.sh_ and _qflow.sh.in_ weren't reflected also. 
[image]

I've taken a few snippets of the debugging process and hope it sheds some light. I will try to roll back to an older version and see if that's stable.  I've used map9v3.v file as instructed in the tutorial so I haven't included that. I also deleted and reinstalled Magic and Qflow a couple of times over from the git repo through the GNU Autoconf.

For those who don't know, the following files can be found in  _qflow/scripts_ and the exec file in the project root directory.

**QflowShellLn394**
[image]

**QflowShellLn880**
[image]

**ExecScriptUpdated**
[image]

**QflowManagerLn1539**
[image]

**QflowManagerLn3239**
[image]

**The _magic_db.sh_ was also checked and added a test line but it wasn't reflected.**
[image]

Thanks for your time

PiyushSaini97, 2021-07-03:
Please refer to this
https://www.gitmemory.com/issue/RTimothyEdwards/qflow/22/812823139

You have to make the changes to all four magic files.
Let me know if this works.

ThimayaPR, 2021-07-03:
Yes, i found the post right after i commented here, but even after changing the files, it didn't work. tried gedit , vim, and chmod to make it executable for a regular user also but they dint work. Then i attempted a reinstall and that too dint work. I rolled back to an older release and now it's fine, have a few LVS errors with my personal files now need to look into it.  [Archive](http://opencircuitdesign.com/qflow/archive/) used 1.4.90 . 

Did editing the shell files work for you? if so how did you do it?

PiyushSaini97, 2021-07-03:
Yes it did work for me. I had to add just an '=' as stated in above post in four files: magic_db.sh, magic_drc.sh, magic_gds.sh and magic_view.sh. 
You can use "find" option available in gedit to find `set techfile ${techdir}/${techfile}` and replace it with 
`set techfile=${techdir}/${techfile}` in all four files.

Since its working, I'll close the issue now.

closed 2021-07-03 by PiyushSaini97