RTimothyEdwards/qflow #27: RePlAce is not detected

minhchuvolam, 2021-06-24:
The `RePlAce` module  is not detected even though it is installed at `/usr/local/bin/replace`.

```
Configuration results:
----------------------------------------------------

Using yosys verilog synthesis tool at: /usr/local/bin/yosys
Using graywolf placement tool at: /usr/local/bin/graywolf
RePlAce not found.  Use --with-replace=<DIR> (optional)
Using qrouter detail route tool at: /usr/local/bin/qrouter
Using Magic layout tool at: /usr/local/bin/magic
Using Netgen LVS tool at: /usr/local/bin/netgen
Using Vesta STA tool (internal)
Using OpenTimer static timing analysys tool at: /usr/local/bin/ot-shell
Using OpenSTA static timing analysys tool at: /usr/local/bin/sta
----------------------------------------------------
```

lootr5858, 2021-09-30:
assuming that you build your replace using the standalone git ...
make a copy of your replace file in <replace_root_dir>/build and rename it to RePlAce
this works for me

lootr5858, 2021-09-30:
if i add RePlAce file inside /usr/local/bin, ntuplace3 and ntuplace4h are not detect
so i use --with-replace=<replace_root_dir>/RePlAce/build/

see my partial output below
`checking for RePlAce... /home/apexlinux/software/RePlAce/build//RePlAce
checking for ntuplace3... /home/apexlinux/software/RePlAce/build//../ntuplace/ntuplace3
checking for ntuplace4h... /home/apexlinux/software/RePlAce/build//../ntuplace/ntuplace4h
configure: creating ./config.status
config.status: creating Makefile
config.status: creating scripts/Makefile
config.status: creating src/Makefile
config.status: creating tech/Makefile
config.status: creating tech/osu050/Makefile
config.status: creating tech/osu035/Makefile
config.status: creating tech/osu035_redm4/Makefile
config.status: creating tech/osu018/Makefile
config.status: creating tech/gscl45nm/Makefile

Configuration results:
----------------------------------------------------

Using yosys verilog synthesis tool at: /usr/bin/yosys
Using graywolf placement tool at: /usr/local/bin/graywolf
Using RePlAce placement tool at: /home/apexlinux/software/RePlAce/build//RePlAce
Using qrouter detail route tool at: /usr/local/bin/qrouter
Using Magic layout tool at: /usr/local/bin/magic
Using Netgen LVS tool at: /usr/local/bin/netgen
Using Vesta STA tool (internal)
Using OpenTimer static timing analysys tool at: /usr/local/bin/ot-shell
Using OpenSTA static timing analysys tool at: /usr/local/bin/sta
----------------------------------------------------
All tools needed for complete flow have been found.
Run "make" followed by "sudo make install" to install qflow.`