RTimothyEdwards/qflow #46: Error is Qflow synthesis

santosh2407, 2023-02-06:
I have installed all the required packages and tools required for running Qflow but after opening the gui and uploading the verilog file, the first step "Preparation" is running successfully when I'm running the synthesis process it is throwing an error saying **"errors encountered in synthesis flow", "Qflow synthesis failed to generate and output  .v file."**

Please help me out!
[image]

Nandinina, 2024-03-21:
Even I have the same issue  please u found solution let me know

santosh2407, 2024-05-25:
This is due to the potential errors in the Verilog file. Please make sure that the verilog file simulated in Linux Environment. You can use iVerilog for the same.

closed 2024-05-25 by santosh2407

Chay702, 2024-09-22:
I am having error in synthesis process 
Qflow synthesis failed to generate an output .blif file